OnionIoT/i2c-exp-driver #3: Image skewed by OLEC Expansion Control of Control Panel after b236

blurfl, 2015-12-09:
Just updated from b234 to b236, now the Image function of the OLED Expansion Control of the Control Panel displays the image in a torn, skewed manner. It worked correctly in b234. Worked correctly in b234.

mccollam, 2015-12-09:
I can confirm this.  Given 
[image]
this is the
[image]

Previously there was no skew.

greenbreakfast, 2015-12-09:
`oledDraw` function now resets the column addressing to the full width (0-127), fixes the skew issue.

Fixed in commit db00176dd41215878c81adfaa1b6ebf58e0ddeb1
Available in Omega firmware b238

mccollam, 2015-12-09:
This fixed the skew issue but there appears to be a few pixels of shifting on the vertical axis (i.e. the bottom few rows of pixels of images are being shifted to the top of the image).

It's difficult to see, but present.  Given the input:
[image]

I get output like:
[image]

(Note that the timer display from the bottom has been partially cut off and pushed onto the top.)

greenbreakfast, 2015-12-09:
Ok, I think I know what's wrong. I'll most likely have a fix sometime tomorrow!

greenbreakfast, 2015-12-19:
Ok, I have a fix for the wrap-around issue. 

Please pickup b246 and try it out. Run `oupgrade -l -f` to grab the firmware.
Let me know how it goes, I've done some testing on my end and it looks ok:
[image]
[image]

I'll close the issue when you guys confirm it's all good on your end :)

blurfl, 2015-12-20:
After updating (both methods) the oled-exp draw command leaves the display blank for me. All other commands work as expected
 Using the Console OLED control I can load an image, but when scrolling 45° the bottom row of pixels seem to be left behind.

greenbreakfast, 2015-12-24:
For the draw command, what kind of file are you pointing to? It's a pretty specific format of data, we're working on a tool to convert images and store them on the omega.

For the second issue, can you try picking up b253 by running `oupgrade -l -f` and trying again? The bottom row of pixels now scrolls on my side. Let me know!

blurfl, 2015-12-24:
For the draw command, I tried .gif, .jpg, .png and .bmp - all these files work using the web page interface, and all worked with the earlier version. 
I'll try the b253 version soon and report back.

blurfl, 2015-12-25:
I've run upgrade -l -f and still find that the  bottom row behaves the same. The old-exp draw command seems to blank evert pixel.

To see what I'm talking about, load the attached picture using the web page and choose 45-degree right scroll. On my setup, the bottom row of pixels, white, remain white while the rest of the image scrolls While that is on the display, use the command line to draw the same image. On my setup, the old blanks except for one pixel that continues to scroll. Repeating that command blanks that pixel as well.

I've attached the file as well as a .zip of it in hopes that will help.

All that said, let me take a moment to thank you for all the work you've put into this platform. The things you add are what will make it truly usable. Have a good holiday, and I hope a long one, as well.

Scott

greenbreakfast, 2015-12-25:
Can you provide a link to the image? The GitHub issues page and the emails don't have any attachments.

You're welcome! It's great to hear that people are enjoying using it and that there's excitement for the future. Happy holidays! :)

-Lazar

greenbreakfast, 2016-01-19:
Hi Scott,
Are you still seeing this issue?

blurfl, 2016-01-19:
Sorry Lazar,
 I’ve been pulled away to other items, unable to pick up and check at this time.

 I’ve attached a .png which should show the issue if it is still present. It’s a 128x64 rectangle, 1 pixel wide, filled with black. Load the picture and scroll; the white row of pixels at the bottom would not go black.

 Sorry to drop out of sight and thanks for all the work you’re doing on the Omega,

Scott

mccollam, 2016-02-13:
Coming back to this very late, but I just checked with firmware 0.0.6 b266 and the issue seems to be resolved.  I am able to write text and images to the OLED screen with no skewing or garbage and clearing and scrolling work correctly.  Thanks @greenbreakfast!

greenbreakfast, 2016-04-19:
no problem, have fun! :)

closed 2016-04-19 by greenbreakfast