OnionIoT/i2c-exp-driver #13: Omega I2C driver misses reading without defining register address

foxel, 2017-03-04:
Currently there is on readBytes(deviceAddr, addr, numBytes) -> [values] method.
It makes impossible to implement some I2C chip protocols. e.g. https://s3.amazonaws.com/controleverything.media/controleverything/Production%20Run%205/10_SI7006-A20_I2CS/Datasheets/Si7006-A20.pdf

There should be something like read(deviceAddr: int, numBytes: int) -> [values] or read(deviceAddr: int, [args]: int[], numBytes: int) -> [values]

mbainter, 2017-04-02:
Yes, this has been a rather frustrating discovery with the [SI7020-A20](https://s3.amazonaws.com/controleverything.media/controleverything/Production%20Run%205/12_SI7020-A20_I2CS/Datasheets/Si7020-A20.pdf) Temp/Humidity CE module as well.

foxel, 2017-04-26:
I've successfully modified the code and created PR.

Compiled packages are available on https://www.dropbox.com/sh/p7mki5nbgyu1tcv/AABPC1HzA848MbEA9RpNiNNva?dl=0

One can install them with `opkg install --force-checksum libonioni2c_0.4-1_mipsel_24kc.ipk pyOnionI2C_0.4-1_mipsel_24kc.ipk pyOmegaExpansion_0.4-1_mipsel_24kc.ipk`

PR #15 (2017-07-26): Fixed #13: Omega I2C reading without defining register address
Fixed #13 (Omega I2C driver misses reading without defining register address)
commit 1f4ee1d5 (2017-07-26): Fixed #13 (Omega I2C driver misses reading without defining register address)
commit 4d6d94b2 (2017-07-26): Merge pull request #15 from foxel/master

Fixed #13: Omega I2C reading without defining register address
closed 2017-07-26 by greenbreakfast