OnionIoT/i2c-exp-driver #17: Error  Error relocating /usr/lib/python2.7/OmegaExpansion/onionI2C.so: i2c_writeBufferRaw: symbol not found"

Cu-Toof, 2017-09-30:
I'm connecting the LCD Display with Onion omega. But i have a issue:

"ImportError: Error relocating /usr/lib/python2.7/OmegaExpansion/onionI2C.so: i2c_writeBufferRaw: symbol not found"

Could you please tell me a solution for my issue, thank you!

alessiodionisi, 2017-09-30:
Same error

RiccardoTeodori, 2017-09-30:
I'm having the same problem with "onionI2C.so" but not with "pwmExp.so" which i think uses the same functions to control the PWM Expansion. The latter doesn't throw errors however and works as expected.

greenbreakfast, 2017-10-02:
Ah I recently released a new version of the I2C library that implements a new function (reading without specifying an address). It looks like your Python library got updated but the C library itself didn't. 

Try running this:
```
opkg update
opkg upgrade libonioni2c pyOnionI2C
```

Let me know if it works!

RiccardoTeodori, 2017-10-04:
I just upgraded the C library (which was on version 0.4-1 instead of 0.5-1) and now the script which used to throw the error doesn't do it anymore. The problem seems solved but I'll do some extensive tests and try to read/write from/to some i2c devices. Thanks sir!