OnionIoT/i2c-exp-driver #20: I2C driver causing segmentation fault in onion-i2c.c

shaheenm82, 2017-10-20:
When performing reads and writes the buffer is often cleared by performing a memset.  the driver assumes a buffer of 32 bytes and always performs 
memset( buffer, 0, I2C_BUFFER_SIZE );

If the user supplies a smaller or bigger buffer, this could not have the intended effect.  For a smaller buffer this can overwrite memory past the end of the supplied buffer, thus causing a segmentation fault.

Rather use the nbytes parameter and only perform a memset on the length of the buffer provided.

greenbreakfast, 2017-10-23:
Yes, we had seen this as well and fixed it in the `read-fix` branch. It just never got brought to the master branch.

Did it now, will release an updated package soon!

maxgerhardt, 2018-05-10:
Is this true? Seems like it was **never** merged. There are still offending lines: 

https://github.com/OnionIoT/i2c-exp-driver/blob/master/src/lib/onion-i2c.c#L161
https://github.com/OnionIoT/i2c-exp-driver/blob/master/src/lib/onion-i2c.c#L313

Re-discovered these while trying to write a I2C device driver and wondering why my programm keeps crashing with a corrupted stack.

greenbreakfast, 2018-06-13:
The `read-fix` branch was indeed merged but it only had fixes for the `i2c_write()` and `i2c_readByte()` functions. This function, `i2c_readRaw()`, had the same issue but was overlooked.

In any case, it's fixed now and part of the v0.7 libonioni2c release. Please try it out and let me know if you have any issues.

maxgerhardt, 2018-06-13:
The code looks good now. I was able to remove my workarounds for library bugs and it does work with the newest version! Issue can be closed IMO.

closed 2018-06-13 by greenbreakfast