OnionIoT/i2c-exp-driver #21: setCursor() function in Python module returns "I2C transaction failed."

impactful, 2017-11-21:
After installing gcc, make, git, and python-dev on Omega2+ booting from SD card, the Python "setCursor()" function fails, with the following message:

> IOError: I2C transaction failed.

The command-line and C versions set row correctly, but not column.  No error is returned.

greenbreakfast, 2017-11-21:
what column are you trying to set the cursor to? have you seen the **set column addressing** functions? https://docs.onion.io/omega2-docs/oled-expansion-python-module.html#oled-py-set-column-addressing

impactful, 2017-11-21:
I have not changed column addressing values from defaults (0 and 127).  

At the command line, a concatenated oled-exp command works:

> oled-exp cursor 3,0 write "How are you?"

But, issuing two consecutive commands:

> oled-exp cursor 3,0 
> oled-exp write "How are you?"

...results in writing starting on the _first_ column.

I'm not sure how to handle this in Python or C.

My work-around has been to issue I2C commands directly.  In Python, this works:

> i2c.writeBytes(0x3C, 0x80, 0xB0) sets cursor to beginning of row 0.
> i2c.writeBytes(0x3C, 0x80, 0xB1) sets cursor to beginning of row 1.
> etc.

greenbreakfast, 2017-11-21:
The `oled-exp write "How are you?"` command will reset the cursor to the very first column. 
Ok, we'll put this on the list of things to debug!