OnionIoT/i2c-exp-driver #23: Off-By-One Error in i2c_writeBuffer

maxgerhardt, 2018-05-10:
I want to use the function `i2c_writeBuffer(int devNum, int devAddr, int addr, uint8_t *buffer, int size)` to talk to an I2C device. However, the function copies one-byte too much into the `malloc`-ed buffer, thus causing a potential heap corruption and other problems. 

```cpp
int i2c_writeBuffer(int devNum, int devAddr, int addr, uint8_t *buffer, int size)
{
	int 	status;
	uint8_t *bufferNew;

	// allocate the new buffer
	size++;		// adding addr to buffer
	bufferNew 	= malloc( size * sizeof *bufferNew );

	// add the address to the data buffer
	bufferNew[0]	= addr;
	memcpy( &bufferNew[1], &buffer[0], size * sizeof *buffer );

 	// perform the write
 	status 	= _i2c_writeBuffer(devNum, devAddr, bufferNew, size);

 	// free the allocated memory
 	free(bufferNew);

	return (status);
}
```

Say the function is called with `size = 3`. Then, `size` will be incremented by one to `4` and `bufferNew` will be a 4-byte buffer on the heap. The first byte is filled with the address. But then, `memcpy( &bufferNew[1], &buffer[0], size * sizeof *buffer );` will again copy `size = 4` bytes starting at `&bufferNew[1]` into the buffer. This is an off-by-one, since it goes 1 byte beyond the size of the buffer. It ought to copy `size - 1` bytes (the original value of the `size` parameter) instead.

zdethcore, 2018-05-21:
(thanks for pointing me to this from the Omega forums) 

good catch on this increasing buffer size, but, i *think* it doesn't affect the intended functionality once in _i2c_writeBuffer.  So, inside of i2c_writeBuffer, in your example, `size` was: 3 | is: 4 and `bufferNew` is 5.  Once passed to `_i2c_writeBuffer`, the actual `write` command still uses `size` (which is 4), so you'll be writing the address + 3 bytes.  Only consequence I see, is that you're wasting the time and space to store garbage in bufferNew[5] (or am I wrong?)

````cpp
// generic function to write a buffer to the i2c bus
int _i2c_writeBuffer(int devNum, int devAddr, uint8_t *buffer, int size)
{
	int 	status;
	int 	fd, index;
        ......
	// write to the i2c device
	status = write(fd, buffer, size);
	..........
	return (status);
}`

maxgerhardt, 2018-05-21:
It will correctly write the write number of bytes to the I2C device (address + initially given buffer), right. (However `bufferNew` has 4 byte allocated, not 5, and the overwritten index is `bufferNew[4]`, the 5th byte from the start, but the rest is correct)

However the damage is already done after the `memcpy` beyond the boundaries of `bufferNew` because the allocated / `malloc()`ed space just isn't big enough (by 1 byte). This will cause an overwrite into some other memory structure on the heap or a heap information block, which can cause subsequent program failure. 

Memory / heap corruption might even be exploitable in special settings, i.e. be used for a system takeover by attacking the programs using this I2C library (https://googleprojectzero.blogspot.de/2015/06/what-is-good-memory-corruption.html).

greenbreakfast, 2018-06-13:
@maxgerhardt yep, you were absolutely correct, we were copying by the incremented `size` instead of the actual size.
We've released v0.7 of libonioni2c, please try it out and let us know if there's any issues.

closed 2018-06-13 by greenbreakfast

maxgerhardt, 2018-06-13:
This issue still prevails with the newest version. It results in the same error message when trying to write 65 bytes or more: `onion-i2c:: write issue for register 0x01, errno is 122: Not supported` (thrown in https://github.com/OnionIoT/i2c-exp-driver/blob/master/src/lib/onion-i2c.c#L108). I think this needs changes deeper in the kernel code? The driver just `write()`s to the device file and the I2C device file rejects it.

maxgerhardt, 2018-06-13:
Yep, this off-by-one was fixed.