OnionIoT/i2c-exp-driver #24: Document maximum I2C transfer size of 64 bytes

maxgerhardt, 2018-05-14:
Nowhere in the code is it mentioned that I2C write calls  of up to only 64 bytes are allowed -- all bigger transfers will fail with error message

```
onion-i2c:: write issue for register 0x01, errno is 122: Not supported
```

Debug output from your `liboniondebug` when such a write is requested:

```
onion-i2c:: writing buffer:
        buffer[0]: 0x01
        buffer[1]: 0xab
        buffer[2]: 0xab
        buffer[3]: 0xab
        buffer[4]: 0xab
        buffer[5]: 0xab
        buffer[6]: 0xab
        buffer[7]: 0xab
        buffer[8]: 0xab
        buffer[9]: 0xab
        buffer[10]: 0xab
        buffer[11]: 0xab
        buffer[12]: 0xab
        buffer[13]: 0xab
        buffer[14]: 0xab
        buffer[15]: 0xab
        buffer[16]: 0xab
        buffer[17]: 0xab
        buffer[18]: 0xab
        buffer[19]: 0xab
        buffer[20]: 0xab
        buffer[21]: 0xab
        buffer[22]: 0xab
        buffer[23]: 0xab
        buffer[24]: 0xab
        buffer[25]: 0xab
        buffer[26]: 0xab
        buffer[27]: 0xab
        buffer[28]: 0xab
        buffer[29]: 0xab
        buffer[30]: 0xab
        buffer[31]: 0xab
        buffer[32]: 0xab
        buffer[33]: 0xab
        buffer[34]: 0xab
        buffer[35]: 0xab
        buffer[36]: 0xab
        buffer[37]: 0xab
        buffer[38]: 0xab
        buffer[39]: 0xab
        buffer[40]: 0xab
        buffer[41]: 0xab
        buffer[42]: 0xab
        buffer[43]: 0xab
        buffer[44]: 0xab
        buffer[45]: 0xab
        buffer[46]: 0xab
        buffer[47]: 0xab
        buffer[48]: 0xab
        buffer[49]: 0xab
        buffer[50]: 0xab
        buffer[51]: 0xab
        buffer[52]: 0xab
        buffer[53]: 0xab
        buffer[54]: 0xab
        buffer[55]: 0xab
        buffer[56]: 0xab
        buffer[57]: 0xab
        buffer[58]: 0xab
        buffer[59]: 0xab
        buffer[60]: 0xab
        buffer[61]: 0xab
        buffer[62]: 0xab
        buffer[63]: 0xab
        buffer[64]: 0xab
onion-i2c:: write issue for register 0x01, errno is 122: Not supported
```

The used function was `i2c_writeBufferRaw`.

greenbreakfast, 2018-06-07:
As far as I recall, the limit is 256 bytes in hardware. Can't remember where I saw that tho.
I'll set it to 256 and we can try it out

greenbreakfast, 2018-06-13:
@maxgerhardt Release v0.7 of libonioni2c that includes this change. Please try it out and let me know how it goes

maxgerhardt, 2018-06-13:
This issue still prevails with the newest version. It results in the same error message when trying to write 65 bytes or more: onion-i2c:: write issue for register 0x01, errno is 122: Not supported (thrown in https://github.com/OnionIoT/i2c-exp-driver/blob/master/src/lib/onion-i2c.c#L108). I think this needs changes deeper in the kernel code? The driver just write()s to the device file and the I2C device file rejects it.

maxgerhardt, 2018-08-07:
Per https://onion.io/2bt-i2c-firmware-update/ this should be fixed, but I don't see any new commit here or in the LEDE source repository. Where are the changes for validation?

greenbreakfast, 2018-09-17:
You are correct, firmware b194 and up should have this fixed. See this commit in the Onion LEDE source repo: https://github.com/OnionIoT/source/commit/0ea89f4ec99a6a4c5a86511796fd14277399a9cc

closed 2019-02-08 by maxgerhardt